bluetuith-org/bluetuith #1: arch install error

se7en-x230, 2022-07-08:
go installs but bluetuith command not found and installing from the release 

`error: missing package metadata in Downloads/bluetuith_0.0.1_Linux_x86_64.tar.gz
error: 'Downloads/bluetuith_0.0.1_Linux_x86_64.tar.gz': invalid or corrupted package
`

darkhz, 2022-07-08:
What program are you using to extract the binary from the tar ball? 

I was able to extract the binary using tar:
`tar -xvzf bluetuith_0.0.1_Linux_x86_64.tar.gz`

and by using the Archive Manager application.

As for installing via `go`, ensure that your $HOME/go/bin is in your $PATH.

se7en-x230, 2022-07-08:
I used pacman.

pacman -U bluetuith_0.0.1_Linux_x86_64 
That gives me the missing metadata error.

darkhz, 2022-07-09:
You can't use pacman to extract and install from the archive file, it is merely a tarball. Use tar or any archive program to extract and use it.

Alternatively, since the package is now in the AUR, you can use the following command to install it:
`pacman -S bluetuith-bin`

closed 2022-07-09 by darkhz

se7en-x230, 2022-07-10:
Nice one. Thanks