bluetuith-org/bluetuith #2: See keybindings in the application

tvararu, 2022-07-19:
Hello! This is great. Much easier to use than `bluetoothctl`, and thanks for making it.

It would be helpful if the application included an easy way to see all the keybindings; same table as the one in the README.

Here's a screenshot of how `pulsemixer` does it, triggered by pressing the `?` key, side by side with `bluetuith`:

[image]

darkhz, 2022-07-22:
The keybindings can be viewed by opening the menu (press Ctrl+m). A dedicated help page/dialog, however, will  be more convenient. Thank you for the suggestion, I will implement this feature.

closed 2022-07-23 by darkhz

tvararu, 2022-07-23:
Thanks a lot for this! I'll try it out ASAP :D

commit c15c52c1 (2022-07-23): ui: Implement help dialog popup

Fixes #2