bluetuith-org/bluetuith #3: Error: Could not setup bluez OBEX agent

helcone, 2022-07-31:
Hi I just installed the binarie on a Arch linux set up and I received the error "Error: Could not setup bluez OBEX agent
" with no other output I am running a really small set up using awesomewm tiling window manager so this might due to dependecies but I am having issues finding whihc one should I install.

darkhz, 2022-08-01:
@helcone, first of all, check if your bluetooth service is up and running. 

This error usually pops up when another bluetooth manager or OBEX manager is running (like `blueman` or `obexftp`), make sure these managers are properly closed before launching bluetuith. For blueman, also make sure that its systray icon isn't visible, if it is, click the systray icon, and then click on "Exit".

If you have followed these steps and this issue still persists, I'll look into it.

helcone, 2022-08-01:
O yes you were right sorry for the confusion either way I am going to leave this closed as for documentation for someone who stumble into it, keep the great work is a great app thanks

luistoledo, 2022-08-08:
Hi.
Closing blueman didn't work (I ensured the process isn't running) :(

bluetuith v0.0.6 (instaled via go install)
bluez 5.64-2
dbus-x11 1.14.0-1

```
bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; preset: disabled)
     Active: active (running) since Sun 2022-08-07 23:06:52 CDT; 58min ago
       Docs: man:bluetoothd(8)
   Main PID: 334393 (bluetoothd)
     Status: "Running"
```

5.15.57-2-MANJARO

darkhz, 2022-08-08:
@luistoledo

darkhz, 2022-08-08:
@luistoledo is blueman-applet running? Can you try uninstalling blueman too and see if it works? The blueman-mechanism DBus session may be running, so blueman's OBEX agent _may_ still be registered.

I cannot replicate the issue here. I started blueman, then closed it via the exit option in the blueman systray icon.

luistoledo, 2022-08-08:
thanks for the fast response.
The applet wasn't running on last test.

I found the culprit, it was blueberry! after killing the daemon, then it worked perfectly.
(almost forgot i have it installed, but your last response guide what to search.)
thanks again

closed 2022-08-10 by darkhz

mackeryd, 2023-01-21:
Hello, I have followed the steps as suggested previously in this thread. I am on a fresh install of arch, i can connect to devices but not send and receive files. I get the errors:

Warning: Could not setup bluez OBEX agent
Warning: Send/receive files is disabled

I do not have blueman, or obex FTP installed... any help?

darkhz, 2023-01-22:
Do you have the OBEX daemon(obexd) installed?

mackeryd, 2023-01-22:
Thanks for the response! It is installed but was not running. Once obexd was started the errors went away, but still no option to send files in the menus

mackeryd, 2023-01-22:
The send files button is there now, i am able to recieve files, but sending i get "unable to find service record"

mackeryd, 2023-01-22:
A google search tells me that issue is with the receiving device...it is working. Thank you!