bluetuith-org/bluetuith #6: Authentication aborting

jon4hz, 2022-08-30:
Hi, this project looks super interesting!

I tried to use bluetuith to connect to a flipperzero, which requires a code for pairing, but unfortunately it didn't work. The authentication code was displayed for about a second on the flipper and then the bluetuith returned `Error: Authentication cancelled`. Is pairing with authentication supposed to work?

darkhz, 2022-08-31:
Pairing with authentication works, I use it to pair and connect to my phone and headset.

Can you try pairing and connecting with bluetoothctl? Also if possible, a small video demonstrating the issue would be helpful.

darkhz, 2023-01-21:
This issue will be closed due to inactivity. You can reopen it  the by posting the information I mentioned above.

closed 2023-01-21 by darkhz