bluetuith-org/bluetuith #7: Release binaries not working on Guix System

Kabouik, 2022-09-18:
I tried to package `bluetuith` for Guix System but I am a newbie and it turned out to be a bit more involving than I hoped because some dependencies need to be packaged too. This is as far as I got after using `guix import go github.com/darkhz/bluetuith`:

```scheme
(define-public go-github-com-darkhz-bluetuith
  (package
    (name "go-github-com-darkhz-bluetuith")
    (version "0.1.2")
    (source (origin
              (method git-fetch)
              (uri (git-reference
                    (url "https://github.com/darkhz/bluetuith")
                    (commit (string-append "v" version))))
              (file-name (git-file-name name version))
              (sha256
               (base32
                "0l9wdq5ja7d6c1l4dxy5r33bsfs1rimcri15q76mjdfmvrnwmplj"))))
    (build-system go-build-system)
    (arguments
     '(#:import-path "github.com/darkhz/bluetuith"))
    (propagated-inputs `(("go-golang-org-x-text" ,go-golang-org-x-text)
                         ("go-golang-org-x-term" ,go-golang-org-x-term)
                         ("go-golang-org-x-sys" ,go-golang-org-x-sys)
                         ("go-golang-org-x-crypto" ,go-golang-org-x-crypto)
                         ("go-github-com-rivo-uniseg" ,go-github-com-rivo-uniseg)
                         ("go-github-com-mitchellh-colorstring" ,go-github-com-mitchellh-colorstring)
                         ("go-github-com-mattn-go-runewidth" ,go-github-com-mattn-go-runewidth)
                         ("go-github-com-lucasb-eyer-go-colorful" ,go-github-com-lucasb-eyer-go-colorful)
                         ("go-github-com-gdamore-encoding" ,go-github-com-gdamore-encoding)
                         ("go-golang-org-x-sync" ,go-golang-org-x-sync)
                         ("go-github-com-schollz-progressbar-v3" ,go-github-com-schollz-progressbar-v3)
                         ("go-github-com-pkg-errors" ,go-github-com-pkg-errors)
                         ("go-github-com-mafik-pulseaudio" ,go-github-com-mafik-pulseaudio)
                         ("go-github-com-jnovack-flag" ,go-github-com-jnovack-flag)
                         ("go-github-com-google-uuid" ,go-github-com-google-uuid)
                         ("go-github-com-godbus-dbus-v5" ,go-github-com-godbus-dbus-v5)
                         ("go-github-com-gdamore-tcell-v2" ,go-github-com-gdamore-tcell-v2)
                         ("go-github-com-darkhz-tview" ,go-github-com-darkhz-tview)
                         ("go-github-com-wifx-gonetworkmanager" ,go-github-com-wifx-gonetworkmanager)))
    (home-page "https://github.com/darkhz/bluetuith")
    (synopsis "TUI-based Bluetooth connection manager")
    (description
     "@command{bluetuith} is a TUI-based bluetooth connection manager, which can
interact with bluetooth adapters and devices.  It aims to be a replacement to
most bluetooth managers, like blueman.")
    (license license:expat)))
```
So I wanted to just use one of the prebuilt binaries in the release, but this is what I get:

```
mat@p3-guix~/P/bluetuith $ ./bluetuith                                                                 16:44:41
exec: Failed to execute process '/home/mat/Projects/bluetuith/bluetuith': The file exists and is executable. Check the interpreter or linker?
```

Any ideas what could be wrong? I'm not sure how to check the interpreter or linker.

darkhz, 2022-09-21:
The prebuilt binaries are dynamically linked to /lib/ld-linux.so.2, which exists on most distributions, except for, in this case, GUIX, which packages ld-linux separately. 

To solve this:
- I will have to provide static binaries, the only reason it is dynamically linked is because of an incorrect environment configuration. I will provide it in the next release, upon which you can do a simple ```(url-fetch <path to binary tarball>)``` and replace build instructions with your custom ones, like extracting the tarball and copying files to /bin. The only issue is the handling of multiple architectures.

- Building from source currently does not work, due to one of the dependencies (rivo/uniseg) requiring go v1.19, which is currently not packaged in guix (the last known version is go v1.17). Downgrading the dependencies will be much more complicated, since rivo/tview now depends on the latest rivo/uniseg.

Now for the above import file that you provided, you have to manually add and define modules. The following Scheme code clones and attempts to build the source. This is incredibly hacky, but it gets to the build stage. Save this to a file (for example build.scm) and run `guix build -f build.scm` or `guix package -f build.scm`

(P.S: I referred to this file [hugo.scm](https://git.0x1b.me/mesuix/tree/mesuix/packages/hugo.scm). Also, writing and debugging the build file is a very frustrating process. This surprisingly took me an entire evening to write.)

```scheme
(define-module (guix packages bluetuith)
#:use-module	(guix build utils)
#:use-module  (guix git-download)
#:use-module 	((guix licenses) #:prefix license:)
#:use-module	(guix build-system go)
#:use-module (guix packages)
#:use-module (gnu packages golang)
#:export (setup-gomod-environment))

(define-public go-github-com-wifx-gonetworkmanager
  (package
    (name "go-github-com-wifx-gonetworkmanager")
    (version "0.4.0")
    (source
      (origin
        (method git-fetch)
        (uri (git-reference
               (url "https://github.com/Wifx/gonetworkmanager")
               (commit (string-append "v" version))))
        (file-name (git-file-name name version))
        (sha256
          (base32
            "0rcx01v7875haq02nh8x40qd19l989mqpfwlk52sgw2mzgsmn20l"))))
    (build-system go-build-system)
    (arguments
      '(#:import-path
        "github.com/Wifx/gonetworkmanager"
	 #:tests? #f
      #:phases
         (modify-phases %standard-phases
           (delete 'build))))
    (propagated-inputs
      `(("go-github-com-godbus-dbus-v5"
         ,go-github-com-godbus-dbus-v5)))
    (home-page
      "https://github.com/Wifx/gonetworkmanager")
    (synopsis "gonetworkmanager")
    (description
      "Go D-Bus bindings for NetworkManager 1.16.")
    (license license:expat)))

(define-public go-github-com-darkhz-tview
  (package
    (name "go-github-com-darkhz-tview")
    (version "0.0.0-20220909015013-18ee50337cd4")
    (source
      (origin
        (method git-fetch)
        (uri (git-reference
               (url "https://github.com/darkhz/tview")
               (commit (go-version->git-ref version))))
        (file-name (git-file-name name version))
        (sha256
          (base32
            "1lrb4r4s88idgl2xlbmsw62d5ax1xcjm745l5hand55a3jjiqjl3"))))
    (build-system go-build-system)
    (arguments
      '(#:import-path "github.com/darkhz/tview"
[... 445 lines cut ...]
        ("go-github-com-pkg-errors"
         ,go-github-com-pkg-errors)
        ("go-github-com-mafik-pulseaudio"
         ,go-github-com-mafik-pulseaudio)
        ("go-github-com-jnovack-flag"
         ,go-github-com-jnovack-flag)
        ("go-github-com-google-uuid"
         ,go-github-com-google-uuid)
        ("go-github-com-godbus-dbus-v5"
         ,go-github-com-godbus-dbus-v5)
        ("go-github-com-gdamore-tcell-v2"
         ,go-github-com-gdamore-tcell-v2)
        ("go-github-com-darkhz-tview"
         ,go-github-com-darkhz-tview)
        ("go-github-com-wifx-gonetworkmanager"
         ,go-github-com-wifx-gonetworkmanager)))
    (home-page "https://github.com/darkhz/bluetuith")
    (synopsis "bluetuith")
    (description
      "bluetuith is a TUI-based bluetooth connection manager, which can interact with bluetooth adapters and devices.
It aims to be a replacement to most bluetooth managers, like blueman.")
    (license license:expat)))

go-github-com-darkhz-bluetuith
```

Kabouik, 2022-09-21:
> The prebuilt binaries are dynamically linked to /lib/ld-linux.so.2, which exists on most distributions, except for, in this case, GUIX, which packages ld-linux separately.
> 
> To solve this:
> 
>     * I will have to provide static binaries, the only reason it is dynamically linked is because of an incorrect environment configuration. I will provide it in the next release, upon which you can either fetch and extract the release tarball, and execute the binary or you can write a custom import script that does a simple `(url-fetch <path to binary tarball>)` and runs custom build instructions for extracting the tarball and copying files to /bin. The only issue is the handling of multiple architectures.

Sounds good, thank you very much for looking into it!

> Now for the above import file that you provided, you have to manually add and define modules. The following Scheme code clones and attempts to build the source, but will not finish due to the point made above. This is incredibly hacky, but it gets to the build stage. Save this to a file (for example build.scm) and run `guix build -f build.scm` or `guix package -f build.scm`.
> 
> (P.S: I referred to this file [hugo.scm](https://git.0x1b.me/mesuix/tree/mesuix/packages/hugo.scm). Also, writing and debugging the build file is a very frustrating process. This surprisingly took me an entire evening to write.)

Yeah, the package definition I posted above was just a first step and most of the work was done by `guix import`. I am not experienced in Guix and or the Scheme language, and I have only submitted a handful of packages so far, it was indeed involving (especially as I am not a developer, so understanding the basics of Scheme was a steep and rocky road).

Thank you so much for spending so much time on it, I did not imagine that you would. If you can get to a working state, then it would definitely be worth submitting the package to Guix. I believe that multiplatform support is recommended but not mandatory for packages.

> ```scheme
> (define-module (guix packages bluetuith)
> #:use-module	(guix build utils)
> #:use-module  (guix git-download)
> #:use-module 	((guix licenses) #:prefix license:)
> #:use-module	(guix build-system [...]
Thanks a ton for that. Now I don't know if I should wait for the next release and extract the tarball (it would work on my machine but I believe this is not a recommended way for Guix packages, not sure that this would pass the review stage), or install it that way (in which case I think packages should not necessarily go into the same .scm file, but I'd need to check how Go packages are organized in Guix upstream at the moment).

darkhz, 2022-09-22:
Hmm, if at least go1.18 is released, this should build properly. I've been experimenting with this some more, and a lot more other convoluted errors pop up, with cryptic stack traces and other errors (the latest was a permission denied error for mkdir). Highly frustrating, that such a simple build process should be made so difficult. Quite the headbanger.

What's funny is, directly cloning and building the source yourself manually (i.e without the import script) doesn't involve much hassle. 

BTW, this release (v0.1.3) contains static binaries, you can easily download it and run on your machine. Do ensure that the required dependencies are installed. I haven't accounted for non-go related deps in the import script (dbus and bluez) yet. I'll try again during the weekend.

> Now I don't know if I should wait for the next release and extract the tarball (it would work on my machine but I believe this is not a recommended way for Guix packages

Can you link the rule where it specifies this? I'd like to go through the packaging guidelines as well.

Kabouik, 2022-09-22:
I can confirm the prebuilt binary works fine on my end, this is awesome (I really love `bluetuith` and even find it surprising that no Bluetooth manager TUI was attempted before, it really fills an empty niche). I didn't have to install anything, I supposed I had all the dependencies already.

I couldn't find any proper rule in the manual, but I think it's just that Guix is a source-based distributions with very few exceptions (and likely not for non-core things), just got confirmation on their IRC channel from some core contributors/maintainers. They agree, though, that building Go programs for Guix is very involving and can be frustrating.

attila-lendvai, 2022-09-22:
FWIW, there are examples for packaging prebuilt binaries for guix here: https://github.com/attila-lendvai/guix-crypto

note that right now this requires the nonguix channel. the readme discusses the situation.

darkhz, 2022-09-23:
@attila-lendvai thank you for the github repo link, it will be a useful reference. I had hoped to publish this into the main guix channel, but as @Kabouik acknowledges, the build process is too involving. I will attempt a rewrite of the build instructions to fetch and install binaries from the github repo (will use [this](https://github.com/attila-lendvai/guix-crypto/blob/f2874ee768c25e972c6f47c581c2f57688e6e47d/src/guix-crypto/packages/ethereum.scm#L110) as reference) and will post an update here once I've tested it.

darkhz, 2022-09-24:
@Kabouik ok here you go, this should work. Possibly review and make the changes that I've highlighted in the comments, if necessary.

```scheme
(define-module (guix packages bluetuith)
	#:use-module  (guix packages)
	#:use-module  (guix download)
	#:use-module  (guix licenses)
	#:use-module  (gnu packages)
	#:use-module  (gnu packages compression)
	#:use-module (nonguix build-system binary))

(define-public bluetuith
  ;; A match statement here to determine the system architecture
  ;; and set the arch variable. Could use %current-system to determine this.
  (let ((arch "x86_64"))	       
  (package
    (name "bluetuith-binary")
    (version "0.1.3")
    (source (origin
              (method url-fetch)
              (uri (string-append "https://github.com/darkhz/bluetuith/releases/download/v" version
				       "/bluetuith_" version "_Linux_" arch ".tar.gz"))
              (sha256
               (base32
                "0iflblpnn4iiqpwkcc7jg08bvjygdz32p1fs9wg43ccpxn4ihrd5"))))
    (build-system binary-build-system)
    (arguments
	(list
	;; The "bin/" directory will be present under a specific build path
	;; under /gnu/store, and the binary will be extracted here, but it
	;; won't be accessible since the path isn't specified in the $PATH variable.
	;; Change "bin/" to preferably the user's local guix-profile path.
        #:install-plan ''(("bluetuith" "bin/"))
        #:strip-binaries? #true))
    (home-page "https://github.com/darkhz/bluetuith")
    (synopsis "bluetuith")
    (description
     "bluetuith is a TUI-based bluetooth connection manager, which can interact with
bluetooth adapters and devices.  It aims to be a replacement to most bluetooth
managers, like blueman.")
    (license expat))))

bluetuith
```

closed 2023-09-02 by darkhz

Kabouik, 2023-09-02:
I can confirk your package definition works @darkhz, thanks a lot for  it. I have been using it for some time now, but haven't found the time to submit it to the nonguix channel because I lack experience with the process (the guix channel would be best for visibility, but as discussed earlier, it would need the package definition to compile `bluetuith` instead of using a pre-compiled binary, and that's not trivial for Go as far as I underatand).

darkhz, 2023-09-02:
@Kabouik yes, for GUIX at least, creating a package definition to compile from source is pretty non-trivial and a very time consuming process, which requires knowledge of Lisp and GUIX related functions as well.

Alternatively, you could perhaps create a post on a GUIX forum(if it exists) and ask for more experienced members to create a package definitions. Perhaps you could link this issue as well.

Kabouik, 2023-09-02:
The most active channel for Guix is #guix on the Libera IRC server, and they're very helpful there (there's also the issue tracker but this is more dedicated to package definitions already quite advanced), but given my knowledge in Scheme/Lisp and Guix, packaging a Go program is kind of an uphill battle and I may waste others' time without succeeding in the end. Hopefully your package with the prebuilt binary can end in nonguix channel at some point and attact the attention of more experienced users.

Kabouik, 2023-11-21:
Just wanted to let you know that after a year of laziness, I finally edited your package definition to submit it to the `nonguix` channel:

```scheme
(define-module (bluetuith)
  #:use-module (gnu packages compression)
  #:use-module (gnu packages glib)
  #:use-module (gnu packages linux)
  #:use-module (guix download)
  #:use-module ((guix licenses) #:prefix license:)
  #:use-module (guix packages)
  #:use-module (ice-9 match)
  #:use-module (nonguix build-system binary))

(define-public bluetuith
  (package
    (name "bluetuith")
    (version "0.1.9")
    (source
     (origin
       (method url-fetch)
       (uri (string-append "https://github.com/darkhz/bluetuith/releases/download/v"
                            version
                            "/bluetuith_" version
                            (match (%current-system)
                              ("x86_64-linux" "_Linux_x86_64")
                              ("i686-linux" "_Linux_i686")
                              ("aarch64-linux" "_Linux_arm64")
                              ("armhf-linux" "_Linux_armv7"))
                            ".tar.gz"))
        (sha256
         (base32
          (match (%current-system)
            ("x86_64-linux" "04ssnbbr2nwl89ygkp777317v9720slwmpdmkfl9grzmq6qbndqw")
            ("i686-linux" "04ssnbbr2nwl89ygkp777317v9720slwmpdmkfl9grzmq6qbndqw")
            ("aarch64-linux" "0h1hynihinvrpf1l67d5yxhrfhiq3x4xi84f3fzfcbwndd7lfsl9")
            ("armhf-linux" "04ssnbbr2nwl89ygkp777317v9720slwmpdmkfl9grzmq6qbndqw"))))))
    (supported-systems '("x86_64-linux" "i686-linux" "aarch64-linux" "armhf-linux"))
    (build-system binary-build-system)
    (arguments
	(list
         #:install-plan ''(("bluetuith" "bin/"))
	 #:strip-binaries? #t))
    (propagated-inputs bluez)
    (synopsis "TUI-based Bluetooth connection manager")
    (description
     "@command{bluetuith} is a TUI-based Bluetooth connection manager, which
can interact with Bluetooth adapters and devices.  It aims to be a replacement
to most Bluetooth managers, like @command{blueman}.")
    (home-page "https://github.com/darkhz/bluetuith")
    (license license:expat)))
```

I have a couple other packages to finish first before submitting it, but I hope to do that later this month. 

I got the OBEX warning when running it, but I only tried it on my Droidian phone so far where it might not be easy to stop the builtin Bluetooth manager; will try on my Guix computer soon.

podiki, 2023-11-21:
The Nonguix issue in question: https://gitlab.com/nonguix/nonguix/-/merge_requests/365

With this building from source it should be submitted to Guix itself. It is a pretty manageable amount of packages and pretty much just straight from the go importer in Guix. The issue of a bluez package input I didn't look at, just wanted to see if I could get it to build (and it even runs) quickly.