bluetuith-org/bluetuith #8: Quit Menu

arghyadipchak, 2022-10-07:
Wouldn't it much easier if it would quit as soon as I pressed 'y' instead of having to type 'y' and then press enter?

darkhz, 2022-10-07:
I suppose, and it can be implemented, but I've designed it that way to prevent accidentally quitting the application. I think pressing <kbd>y</kbd>+<kbd>Enter</kbd> provides a **somewhat** more (granular?) control over the quitting process, compared to just pressing <kbd>y</kbd>. I'm open to change this though.

arghyadipchak, 2022-10-07:
I think just `y` would be fine as the person is already pressing `q` because he wishes to quit.
To prevent accidental quitting maybe you can make the prompt like `Quit (y/N) ? ` so if the person presses anything other than `y`, `N` is taken as default so it doesn't quit and it only quits if `y` is pressed.

darkhz, 2022-10-08:
I'll consider implementing this

carsin, 2022-10-23:
+1, I'd be fine with disabling the quit confirmation altogether

darkhz, 2023-01-21:
> I think just `y` would be fine as the person is already pressing `q` because he wishes to quit. To prevent accidental quitting maybe you can make the prompt like `Quit (y/N) ? ` so if the person [...]

Ok, this will be implemented in the next release.

> +1, I'd be fine with disabling the quit confirmation altogether

Disabling the quit confirmation is somewhat "dangerous" in my opinion. For example, if you are running a send/receive file transfer, and you accidentally press Q, the file will be corrupted, various crashes/errors will occur either on bluez, or the bluetooth stack on the other device where you initiated the transfer, since the bluetooth agent wasn't properly terminated.

xeruf, 2023-01-26:
How about adding Shift+Q i.e. big Q as instant quit option?
This is easy to type but hardly happens accidentally.

katexochen, 2023-06-21:
Faster quitting is definitely a feature I would greatly appreciate. I think shift+q is a valid choice.

> For example, if you are running a send/receive file transfer, and you accidentally press Q

Just to give some context, personally, I use bluetuith as comfortable interface to scan, pair and connect devices. It opens when I click on the bluetooth icon in my waybar (and I would love to be able to quit quickly after connecting). I hadn't a use case for the transfer feature yet. Thanks for developing such a great tool. :+1:

commit 19e98d5d (2023-06-22): ui: menubar: Quit on Shift+q keyevent

Fixes #8
closed 2023-06-22 by darkhz