bluetuith-org/bluetuith #9: Quit on Ctrl-C

xeruf, 2023-01-08:
I am so used to that, always have to search for Q ^^

darkhz, 2023-01-10:
I'll consider implementing this.

darkhz, 2023-01-21:
Do look at #8, and see if the requested feature in that issue will suffice. I blocked Ctrl+C to prevent accidental quitting. Pressing Q+y wouldn't be that tedious, I suppose. If this issue gains traction(i.e if more people request this feature), I can implement it.

closed 2023-06-16 by darkhz

xeruf, 2023-06-22:
Ctrl-C is easier to reach and a habit in the terminal. How would I hit that accidentally? I would really like to have it as an option, too.

darkhz, 2023-06-23:
Hmm, I will look into it. Perhaps if I made keybindings configurable, it would be better for everyone.

xeruf, 2023-06-23:
that was also a thought of mine, but I think that might be unnecessary work
just follow standards - Ctrl-C aborts almost any terminal application, I do not see why this one should be different

darkhz, 2023-06-23:
I will consider this