bluetuith-org/bluetuith #10: Suspension and Disconnect

4iMh9CawHdyu, 2023-01-12:
Works great, better than blueman for the new distros adopting it now; However, just as blueman does, after a while in suspension it disconnects and you have to pair it again..... in my case with a keyboard.

darkhz, 2023-01-14:
Can you elaborate on the suspension? Do you suspend to shell or do you suspend the computer? Because in the former case at least, it does not disconnect.

4iMh9CawHdyu, 2023-01-14:
Suspend the computer, haven't tried suspending to shell...... however, later I found out it had to do with the wifi/bt powersave settings so this command:
sudo sed -i 's/wifi.powersave = 3/wifi.powersave = 2/' /etc/NetworkManager/conf.d/default-wifi-powersave-on.conf

would fix it, then it behaves like blueberry does where it just takes a few seconds after touching something on the keyboard and it is back on..... ONLY PROBLEM IS, after that I cannot connect to the internet anymore. Also there are other issues with Linux Mint 21.1 I've been having after coming back from suspend, so haven't gone too deep there..... like the file manager won't start anymore and after a while the system just stalls. So, for the moment I'm keeping to Mint 20.3 (blueberry, doesn't require turning off powersave and works well... for now. eventually support will be taken off. The change from blueberry to blueman that came along from Mint 20 to 21 has affected a lot of people and bluetuith seems to address that issue.... in Linux MX (also with blueman) I never had this issue and this goes way back to the "el patito feo" distro which i think was number 19 (today it is 21 as well) so they seem to have something worked out with bueman there that Mint hasn't yet. Also, it is Debian..... so that may contribute to the stability of it all.

darkhz, 2023-01-14:
Hmm, this sounds like a kernel/distro issue rather than a problem with the softwares mentioned. I will be closing this issue due to this reason.

On a side note, blueman, I believe, has a behaviour where it automatically rfkills the Bluetooth functionality, possibly on suspend as well. This might be happening on your distro. Do look into it.

closed 2023-01-14 by darkhz

4iMh9CawHdyu, 2023-01-15:
You are right, but, all of the above is precisely the reason why I started to look into your program and why I gave as much information as I currently have.

4iMh9CawHdyu, 2023-01-15:
So, just to update you...... in Linux Mint 21.1 I can only connect my keyboard to Blueman THROUGH Bluetuith...... and once that is done in the plugin section on Blueman I have to disable Killswitch,,,, then everything is solved, after suspend it all works, even the stalling and file manager not showing up that I was telling you about get fixed with this tweak.... just to let you know.
MANY, MANY THANKS!

darkhz, 2023-01-16:
No problem :)