bluetuith-org/bluetuith #16: how can auto scan

newyorkthink, 2023-06-19:
when i enter bluetuith

[image]
is just not scan

and i always enter s ,i scan

[image]

so how can bluetuith start and auto scan driver

darkhz, 2023-06-22:
Currently there is no way to auto-scan on launch, but I can implement this

frankier, 2023-10-24:
I agree it's would be a nice default behavior, possibly with some option to disable

darkhz, 2023-10-25:
This feature has now been implemented with the `adapter-states` command-line option.

To scan automatically on initialization, it can be done via the command-line: 
```
bluetuith --adapter=<adapter> --adapter-states="powered:on, scan:on"`
```

Or by setting the configuration parameter:
```
adapter-states: "powered:on, scan:on"
```

closed 2023-10-25 by darkhz