bluetuith-org/bluetuith #17: new version cannot close

newyorkthink, 2023-06-22:
when i close bluetuith, use q, or ctrl c is not work 
so i just use kill command.

and i kill bluetuith.
it show look this
[image]

cannot run anything

darkhz, 2023-06-22:
The new keybinding is now Shift+q, I forgot to update the docs, I'll do it
soon. See #8.

darkhz, 2023-06-22:
The new keybinding is Shift+q, I forgot to update the docs, I'll do it soon. See #8.

closed 2023-06-22 by darkhz

sabrehagen, 2023-06-23:
@darkhz can this change be added to `invidtui` too please

darkhz, 2023-06-23:
@sabrehagen sure

sabrehagen, 2023-06-24:
@darkhz seeing as no other function is bound to `q`, why not make `q` quit? `q` is common throughout many tui utilities as the method of exit, I have never encountered `Shift+q` before.

darkhz, 2023-06-24:
It's very easy to accidentally press 'q', so I think Ctrl+C may be a more suitable choice.

sabrehagen, 2023-07-11:
I would disagree, for a competent user 'q' is not usually an issue. Writers of `less`, `more`, etc. i.e. unix core tools must not have felt it was an issue.

darkhz, 2023-07-11:
That's true, let me look into this