bluetuith-org/bluetuith #18: Menu close delay

sabrehagen, 2023-06-23:
After opening the menu with `Ctrl+m`, I close the menu with `Esc`. There is a one second delay after pressing `Esc` before the menu closes. What is the cause of this delay? Can it be removed?

darkhz, 2023-06-23:
I can't reproduce this issue on my end, when I press Ctrl+m and Esc the menu closes immediately. Can you record a gif/video of the issue?

sabrehagen, 2023-06-24:
This video shows the delay I observe. Somewhere between 0.5-1.0 sec. Do you observe the same?

https://github.com/darkhz/bluetuith/assets/1094600/05a0ed23-e13f-4637-a2cf-9fb289ca2a27

darkhz, 2023-06-24:
Interesting, I will look into it.

darkhz, 2023-07-02:
Hmm, this may be a very trivial question, but can you perhaps try it in a different terminal emulator? I cannot reproduce it on my end, the menu dialog closes immediately for me

sabrehagen, 2023-07-11:
Yes, testing in xterm has no delay. Delay is experienced in alacritty.

closed 2023-09-02 by darkhz