bluetuith-org/bluetuith #25: Put config file in $XDG_CONFIG_HOME

Anas-Sherif, 2023-07-24:


Anas-Sherif, 2023-07-24:
did it in #26

darkhz, 2023-07-24:
Why would you need to put an application's configuration in the user's home directory? The appropriate place is the .config directory itself.

And in the event the .config directory isn't present,  bluetuith will automatically put its configuration folder and files in your home directory.

Anas-Sherif, 2023-07-24:
the request isn't to put it in the home folder, it's to put it in $XDG_CONFIG_HOME instead of .config (on my system $XDG_CONFIG_HOME isn't .config)

darkhz, 2023-07-24:
Sorry, I misinterpreted the issue, I only skimmed through the description and assumed you meant the home directory. I'll look into merging this.

darkhz, 2023-09-02:
This issue has been fixed in the latest release

closed 2023-09-02 by darkhz