bluetuith-org/bluetuith #27: Keybind quit is set to Shift-q (Q) and not q as the documentation says

kjuulh, 2023-09-02:
Love the tool, I just switched from an older version of bluetuith, and couldn't exit because `q` didn't work. I looked up the documentation which still mention that the key is `q`, however in the menu and source it is set to `Q`. It isn't really a bit deal, I'd be happy to change either if needed =D

https://github.com/darkhz/bluetuith/blob/a088dae22f522fd48ec164ddd57d1e0163bc5f82/cmd/keybindings.go#L103-L108

darkhz, 2023-09-02:
Ah apologies, I forgot to set the correct keybinding in the documentation. You can modify the quit behavior by setting this in your configuration file:

```
{
     keybindings: {
          Quit: q
     }
}
```

kjuulh, 2023-09-03:
No worries. I came from an old aur pkg, so was just a bit confused. I am fine with the change =D

closed 2023-09-03 by kjuulh