bluetuith-org/bluetuith #30: Default quit behavior changed

jack-champagne, 2023-10-02:
Hi - I recently updated from v0.1.3 (the last updated version on the AUR for bluetuith package - not bluetuith-bin or bluetuith-git) up to v0.1.7. I noticed two changes, lowercase 'q' does not quit the application anymore. Additionally, when I do quit the application (by pressing uppercase "Q"), I am not prompted if I really want to quit anymore.

I think that lowercase 'q' should still act as a quit key. If there are good reasons for this _not_ to happen then the help menu should be updated to reflect that uppercase "Q" is the correct key to Quit

Also since users are not prompted anymore on quit (to confirm quit), either that functionality should be added back or the confirmQuit func should be removed from the code (I believe it isn't called anywhere else).

I am happy to make the PR for these changes if they make sense. Thank you for making such a great tui for bluetooth!

darkhz, 2023-10-05:
Do look at #8 to see why I removed the quit confirmation. I am open to adding it back, but I cannot ignore said issue, so I am frankly not sure what to do.

As to the confirmQuit function, yes I think that must be deleted, I will have a look into it, thank you for pointing that out.

As for the quit key, as of v0.1.7 it is now modifiable via the configuration file. So to modify the keybinding to 'q', you would add the following directive to the configuration:

```
keybindings: {
Quit: q
}
```

darkhz, 2023-10-21:
Maybe a confirm-on-quit commandline/config option might help? I will look into implementing this, but submit your PR anyway, I will review that as well.

darkhz, 2023-10-25:
This feature has now been implemented with the `confirm-on-quit` command-line option.

To show a confirmation message before quitting the application, it can be done via the command-line:
```
bluetuith --confirm-on-quit
```

Or by setting a configuration parameter:
```
confirm-on-quit: true
```

closed 2023-10-25 by darkhz