bluetuith-org/bluetuith #31: Change navigation keys?

hacktail, 2023-10-15:
Hi, is there some way to change the navigation keybindings? I've got a Nordic keyboard setup which means that the standard `h j k l` has to be replaced into `j k l ö`, as it will otherwise defeat the purpose of not using other forms of navigation.

darkhz, 2023-10-15:
There is currently no way to change the navigation keybindings, but it can be implemented.

hacktail, 2023-10-16:
> There is currently no way to change the navigation keybindings, but it can be implemented.

W

hacktail, 2023-10-16:
- [ ] 

> There is currently no way to change the navigation keybindings, but it can be implemented.

I would be extremely grateful if you did, can be a deal breaker sometimes.

darkhz, 2023-10-16:
Okay, this will be implemented in the next release

darkhz, 2023-10-25:
This has now been implemented in this release.

To change the navigation keys, it can be done by setting configuration parameters:
```
keybindings {
    # Similar to pressing the 'Up' key
    NavigateUp: 'k'

    # Similar to pressing the 'Down' key
    NavigateDown: 'l'

    # Similar to pressing the 'Right' key
    NavigateRight: 'j'

    # Similar to pressing the 'Left' key
    NavigateLeft: 'ö'
}
```

closed 2023-10-25 by darkhz

hacktail, 2023-11-06:
> This has now been implemented in this release.
>
> To change the navigation keys, it can be done by setting configuration parameters:
>
> ```
> keybindings {
>     # Similar to pressing the 'Up' key [...]

Thank you so much! Maybe I should've made my own PR, maybe I'll do that next time :)