bluetuith-org/bluetuith #32: Pairing keyboard - increase display time of the access number

jhmj-io, 2023-10-20:
Whilst pairing a keyboard (MX keys) entering a code on the keyboard is customary. This code is displayed in the bluetuith interface - display time is very short though.

darkhz, 2023-10-20:
Are you on the latest version? Can you perhaps send a screenshot of the code display that you are getting?

jhmj-io, 2023-10-20:
yes, I'm on the latest version 

./bluetuith --version says: Bluetuith v[]

[image]

darkhz, 2023-10-20:
Okay, from the screenshot it does look like it is the latest version. I think you must have compiled the program yourself, which is why the correct version is not displaying properly.

I will look into this.

jhmj-io, 2023-10-20:
yes I compiled myself - Fedora

darkhz, 2023-10-21:
@jhmj-io  Just to confirm, the message displayed is "Passkey for MX keys is 123456, entered ------"?

jhmj-io, 2023-10-21:
YESS confirmed!
[image]

darkhz, 2023-10-25:
This feature has now been implemented in this release. All pairing codes and requests will now be shown indefinitely within a popup, and you can close the popup by either pressing `Escape` or clicking the 'X' button.

closed 2023-10-25 by darkhz