bluetuith-org/bluetuith #33: Nano-style on-screen help by default

frankier, 2023-10-24:
Thanks a lot for the project. It's already at the top end of Linux bluetooth interfaces I have used in terms of comfort for me. I wonder whether it would be possible to display the keybindings by default like nano does? An equivalent of `-x` and `set nohelp` could be added.

darkhz, 2023-10-25:
It can be implemented, but my only concern is that is could eat up screen space, especially since there are a lot of keybindings for a single screen. Wouldn't pressing the `Help` key suffice?

frankier, 2023-10-25:
For me, I am occasionally using this software and so will never remember the keybindings. Of course if you prefer to put the bindings behind a help key that's still more convenient than reading documentation, but while it is very much your choice, I would make attempt to make an argument that the common user is a less frequent users and would thus be best served by the nano-style on-screen help that is easily hidable.

darkhz, 2023-10-25:
Fair argument, and I will consider implementing this. As stated before, since there are a lot of keybindings for just one screen, and I do not want to use screen space unnecessarily, this will require some contemplation.

Maybe the main/commonly used keybindings to facilitate basic operations can be suggested to the user for each screen, I will have to check and run some tests. If you have any ideas on how to implement this, preferably with a PR or screenshots, do submit them, as they can be helpful in implementing this feature.

hacktail, 2023-11-06:
I believe nothing more than navigation, connect/disconnect, the menu ( shortcuts? Well the bindings to show the menus ) and ofcourse the help binding should be shown.

The nagivation buttons should be concise as they will otherwise take up a huge amount of space; possibly it could be shown as this: `←/↓/↑/→ =  h/j/k/l` or possibly `←h/↓j/↑k/→l` or maybe replace the arrows with `L D U R` incase we shouldn't use unicode.
maybe we could display the arrows/letters/words under the bindings, like on another row?
```
h j k l
< ^ ˇ >

darkhz, 2023-12-28:
[image]
@frankier @hacktail, how does this look? Would this suffice? A similar set of limited help keybindings will be present on the file picker and the progress view as well.

hacktail, 2023-12-28:
> [image]
> @frankier @hacktail, how does this look? Would this suffice? A similar set of limited help keybindings will be present on the file picker and the progress view as well.

Looks amazing

darkhz, 2023-12-28:
This will be implemented in the next release, and any modifications to the help display can be further suggested by reopening this issue once it is closed.

darkhz, 2023-12-30:
This feature has now been implemented and enabled by default, and can be disabled by:
- Using the command-line option `--no-help-display`
- Setting the configuration parameter `no-help-display: true`

closed 2023-12-30 by darkhz

frankier, 2023-12-30:
Thanks a lot. It looks great!