bluetuith-org/bluetuith #41: Would you port this great tool to FreeBSD

dr3mro, 2023-11-01:
Hello, 
FreeBSD does not have any thing like this great tool, so would you port this tool to FreeBSD
I hope you find this a good point
Thanks

darkhz, 2023-11-03:
You may need to find experienced maintainers who can port this package to FreeBSD, since I haven't worked with it before.

closed 2023-11-03 by darkhz

dr3mro, 2023-11-03:
The issue is freeBSD has no Bluetooth connection manager and users have to
enter multiple commands to initiate a connection
Your work is very creative and freeBSD needs something like this
Linux has a lot of tools already

darkhz, 2023-11-05:
I could perhaps build and publish binaries for FreeBSD, I will do so in the next release.

darkhz, 2023-11-05:
This tool relies on BlueZ to manage bluetooth. I don't think the FreeBSD has the API necessary for this application to work.