bluetuith-org/bluetuith #43: Keyboard navigation

Kabouik, 2023-11-21:
Even though `bluetuith` is built around mouse support, as a TUI, it would be very nice if itcould be fully controlled from the keyboard. Unfortunately, I can't seem to navigate through the three top entries or their contextual menus without clicking, be it using arrows, tab, or vi-bindings. Only up and down arrows seem to allow moving the selection among known devices.

Even without going too complicated, just allowing selecting those entries with arrows, and showing/hiding their contextual menus with Return/Escape would be very useful. A `h` or `?` keybinding to show all available keybindings for the current selection would be handy too, though that would likely require more work?

Kabouik, 2023-11-21:
Okay, so I just found that, which shows this should already be covered: https://darkhz.github.io/bluetuith/Configuration.html
.
I'm just not sure why `Left` and `Right` don't seem to work to select "Adapter" and "Device".

Sorry for the noise.

Kabouik, 2023-11-21:
Alright, I just needed to set a keybinding for `Menu` to be able to navigate through them, the default `Alt+m` was conflicting with my WM's bindings. Somehow, some bindings fail to work even if they don't trigger an error, like `Ctrl+m` (won't even show in the `?` menu), or `Shift+Enter` (will show, but have no effect). I ended up setting `Ctrl+Space`.

closed 2023-11-21 by Kabouik

darkhz, 2023-11-21:
Some keybindings will conflict because they translate to other keybindings, or because the terminal does not capture them. For example, `Ctrl+m` is like pressing the `Enter` key.

`Left` and `Right` are not configured to switch context menus, only `Tab` (or a configured key to cycle through menus) can do that.