bluetuith-org/bluetuith #51: [FEAT] Supporting Vim style navigation and keymap

Krr0ptioN, 2024-01-29:
## Description

Vim style navigation is broadly considered something as general feature for most of TUI application. The existing conventional navigation do not conflict with Vim navigation. So why not?

**NOTE:** I am guessing the maintainer is not a fan of Vim, so that's why the description might seem like a sales pitch :)

darkhz, 2024-01-30:
Take a look at #31 for implementing basic vim-like navigation.

Krr0ptioN, 2024-01-30:
I didn't see that and actually I was about to PR some changes, although it seems like no issue is left to solve.
Repeatable navigational commands could be interesting, e.g. `3j` three times navigate to down. 

Good project though, thanks.

darkhz, 2024-01-30:
I will consider implementing this feature.

closed 2024-02-03 by Krr0ptioN

kohane27, 2024-02-21:
Thanks for adding support for vim-like navigation! I'm just happy with the following:

```
keybindings {
    NavigateUp: 'k'
    NavigateDown: 'j'
    NavigateRight: 'l'
    NavigateLeft: 'h'
}
```

Thank you and have a good day!