bluetuith-org/bluetuith #52: i have idea, use local linux bluetuith remote to  poweroff bluetooth machine

newyorkthink, 2024-02-03:


darkhz, 2024-02-03:
Can you elaborate please?

newyorkthink, 2024-02-03:
> Can you elaborate please?

i mean is: use bluetuith tui or cmd poweroff bluetooth machine like phone bear

darkhz, 2024-02-03:
What you are trying to say is still not clear, but if I assume correctly, you are asking is to power-off a connected Bluetooth device from the TUI?

newyorkthink, 2024-02-03:
> What you are trying to say is still not clear, but if I assume correctly, you are asking is to power-off a connected Bluetooth device from the TUI?

yes , i meaing is around blutuith poweroff Bluetooth device

darkhz, 2024-02-03:
If you mean powering off the host device, bluetuith can already do that, press <kbd>o</kbd> to toggle host controller power states. If you mean switching off the remote device's bluetooth controller, it is not supported.