bluetuith-org/bluetuith #54: Cannot set adapter power state

DavShaw, 2024-02-12:
Hello, I just installed Bluetuith, but when I try to power on de adapter I got that problem... (I cannot scan, pair or connect cuz adapter's powered off)

[image]

darkhz, 2024-02-13:
Hmm, does setting the power for that adapter work while using bluetoothctl or other Bluetooth managers?

darkhz, 2024-02-13:
@DavShaw check whether your bluetooth kernel modules are loaded correctly, and check whether the bluetooth service is running properly with `systemctl status bluetooth.service`, if you use **systemd** for example. Post the journal for the service or relevant `dmesg` logs here.

closed 2024-03-01 by DavShaw