bluetuith-org/bluetuith #55: it always show this, cannnot connect

newyorkthink, 2024-02-23:
[image]

5huanghuai, 2024-03-03:
i have alse problem.
```shell
OS: Arch Linux x86_64
Kernel: 6.7.6-arch1-2

sudo systemctl status bluetooth
===>
===>
Mar 03 12:02:58 Arch bluetoothd[1161]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for xx:xx:xx:xx:xx:xx: Protocol not avai>
```

darkhz, 2024-03-03:
This should be an issue with bluez itself, with the A2DP protocol. Search the Arch forums for how to resolve this issue.

5huanghuai, 2024-03-03:
> This should be an issue with bluez itself, with the A2DP protocol. Search the Arch forums for how to resolve this issue. Maybe check your PipeWire/PulseAudio configuration?

I'm using the default PulseAudio configuration. This problem seems to occur frequently. I've tried many ways to solve it, but I've failed. Eventually, the only solution seems to be using Blueman. However, I really like Bluetooth, so this is embarrassing.

darkhz, 2024-03-03:
Does this issue not occur with blueman? An error like that, along with what the logs say for the bluetoothd service, could originate from some audio implementation related issue, and does seem to happen often on PulseAudio.

darkhz, 2024-03-03:
Maybe, make sure you are running no other Bluetooth managers first, then restart bluetoothd and pulseaudio, and then try connecting again.

darkhz, 2024-03-04:
Okay, I'll look into this. Post the logs (from `journalctl` or `systemctl status bluetooth.service`) here, and maybe try deleting the device and re-pairing it. It may be that you have to re-authorize a specific service.

5huanghuai, 2024-03-07:
This is the log that i uninstall blueman and clear all catche
```shell
● bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; preset: disabled)
     Active: active (running) since Thu 2024-03-07 18:35:27 CST; 3min 20s ago
       Docs: man:bluetoothd(8)
   Main PID: 1668 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 38115)
     Memory: 2.5M (peak: 3.1M)
        CPU: 72ms
     CGroup: /system.slice/bluetooth.service
             └─1668 /usr/lib/bluetooth/bluetoothd

Mar 07 18:35:27 Arch bluetoothd[1668]: profiles/audio/micp.c:micp_init() D-Bus experimental not enabled
Mar 07 18:35:27 Arch bluetoothd[1668]: src/plugin.c:plugin_init() System does not support micp plugin
Mar 07 18:35:27 Arch bluetoothd[1668]: src/plugin.c:plugin_init() System does not support vcp plugin
Mar 07 18:35:27 Arch bluetoothd[1668]: src/plugin.c:plugin_init() System does not support mcp plugin
Mar 07 18:35:27 Arch bluetoothd[1668]: src/plugin.c:plugin_init() System does not support bass plugin
Mar 07 18:35:27 Arch bluetoothd[1668]: src/plugin.c:plugin_init() System does not support bap plugin
Mar 07 18:35:27 Arch bluetoothd[1668]: Bluetooth management interface 1.22 initialized
Mar 07 18:35:27 Arch bluetoothd[1668]: Battery Provider Manager created
Mar 07 18:38:16 Arch bluetoothd[1668]: src/adv_monitor.c:btd_adv_monitor_power_down() Unexpected NULL btd_adv_monitor_manager object upon power down
Mar 07 18:38:42 Arch bluetoothd[1668]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 88:xx:E8:71:xx:38: Protocol not avail>
~
```

This is the log that i reinstall blueman and restart bluetooth 
```shell
sudo systemctl status  bluetooth                                                                                                        18:41:37
● bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; preset: disabled)
     Active: active (running) since Thu 2024-03-07 18:41:37 CST; 6s ago
       Docs: man:bluetoothd(8)
   Main PID: 6637 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 38115)
     Memory: 916.0K (peak: 1.8M)
        CPU: 43ms
     CGroup: /system.slice/bluetooth.service
             └─6637 /usr/lib/bluetooth/bluetoothd

Mar 07 18:41:37 Arch bluetoothd[6637]: Endpoint registered: sender=:1.63 path=/MediaEndpoint/A2DPSink/sbc
Mar 07 18:41:37 Arch bluetoothd[6637]: Endpoint registered: sender=:1.63 path=/MediaEndpoint/A2DPSource/sbc
Mar 07 18:41:37 Arch bluetoothd[6637]: Endpoint registered: sender=:1.63 path=/MediaEndpoint/A2DPSink/sbc_xq_453
Mar 07 18:41:37 Arch bluetoothd[6637]: Endpoint registered: sender=:1.63 path=/MediaEndpoint/A2DPSource/sbc_xq_453
Mar 07 18:41:37 Arch bluetoothd[6637]: Endpoint registered: sender=:1.63 path=/MediaEndpoint/A2DPSink/sbc_xq_512
Mar 07 18:41:37 Arch bluetoothd[6637]: Endpoint registered: sender=:1.63 path=/MediaEndpoint/A2DPSource/sbc_xq_512
Mar 07 18:41:37 Arch bluetoothd[6637]: Endpoint registered: sender=:1.63 path=/MediaEndpoint/A2DPSink/sbc_xq_552
Mar 07 18:41:37 Arch bluetoothd[6637]: Endpoint registered: sender=:1.63 path=/MediaEndpoint/A2DPSource/sbc_xq_552
Mar 07 18:41:37 Arch bluetoothd[6637]: Endpoint registered: sender=:1.63 path=/MediaEndpoint/A2DPSink/faststream
Mar 07 18:41:37 Arch bluetoothd[6637]: Endpoint registered: sender=:1.63 path=/MediaEndpoint/A2DPSource/faststream
```

darkhz, 2024-03-07:
@5huanghuai did you try deleting the device from bluetuith and re-pairing? Also, this seems to indicate a problem with bluetoothd itself. Very weird, however, that it works only when blueman is installed. I'll look into this.

darkhz, 2024-03-07:
Also maybe check section 2.4.4 of https://wiki.archlinux.org/title/bluetooth_headset

5huanghuai, 2024-03-08:
I think that i know where is the problem.
This is the log
```shell
● bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; preset: disabled)
     Active: active (running) since Fri 2024-03-08 09:21:32 CST; 32s ago
       Docs: man:bluetoothd(8)
   Main PID: 1642 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 38115)
     Memory: 2.3M (peak: 2.7M)
        CPU: 38ms
     CGroup: /system.slice/bluetooth.service
             └─1642 /usr/lib/bluetooth/bluetoothd

Mar 08 09:21:32 Arch bluetoothd[1642]: Starting SDP server
Mar 08 09:21:32 Arch bluetoothd[1642]: src/plugin.c:plugin_init() System does not support csip plugin
Mar 08 09:21:32 Arch bluetoothd[1642]: profiles/audio/micp.c:micp_init() D-Bus experimental not enabled
Mar 08 09:21:32 Arch bluetoothd[1642]: src/plugin.c:plugin_init() System does not support micp plugin
Mar 08 09:21:32 Arch bluetoothd[1642]: src/plugin.c:plugin_init() System does not support vcp plugin
Mar 08 09:21:32 Arch bluetoothd[1642]: src/plugin.c:plugin_init() System does not support mcp plugin
Mar 08 09:21:32 Arch bluetoothd[1642]: src/plugin.c:plugin_init() System does not support bass plugin
Mar 08 09:21:32 Arch bluetoothd[1642]: src/plugin.c:plugin_init() System does not support bap plugin
Mar 08 09:21:32 Arch bluetoothd[1642]: Bluetooth management interface 1.22 initialized
Mar 08 09:21:32 Arch bluetoothd[1642]: Battery Provider Manager created
```
And i try `pulseaudio --start`. Surpring things happend

```shell
● bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; preset: disabled)
     Active: active (running) since Fri 2024-03-08 09:21:32 CST; 45s ago
       Docs: man:bluetoothd(8)
   Main PID: 1642 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 38115)
     Memory: 2.3M (peak: 2.8M)
        CPU: 42ms
     CGroup: /system.slice/bluetooth.service
             └─1642 /usr/lib/bluetooth/bluetoothd

Mar 08 09:22:12 Arch bluetoothd[1642]: Endpoint registered: sender=:1.37 path=/MediaEndpoint/A2DPSink/sbc_xq_453
Mar 08 09:22:12 Arch bluetoothd[1642]: Endpoint registered: sender=:1.37 path=/MediaEndpoint/A2DPSource/sbc_xq_453
Mar 08 09:22:12 Arch bluetoothd[1642]: Endpoint registered: sender=:1.37 path=/MediaEndpoint/A2DPSink/sbc_xq_512
Mar 08 09:22:12 Arch bluetoothd[1642]: Endpoint registered: sender=:1.37 path=/MediaEndpoint/A2DPSource/sbc_xq_512
Mar 08 09:22:12 Arch bluetoothd[1642]: Endpoint registered: sender=:1.37 path=/MediaEndpoint/A2DPSink/sbc_xq_552
Mar 08 09:22:12 Arch bluetoothd[1642]: Endpoint registered: sender=:1.37 path=/MediaEndpoint/A2DPSource/sbc_xq_552
Mar 08 09:22:12 Arch bluetoothd[1642]: Endpoint registered: sender=:1.37 path=/MediaEndpoint/A2DPSink/faststream
Mar 08 09:22:12 Arch bluetoothd[1642]: Endpoint registered: sender=:1.37 path=/MediaEndpoint/A2DPSource/faststream
Mar 08 09:22:14 Arch bluetoothd[1642]: Failed to add UUID: Authentication Failed (0x05)
Mar 08 09:22:16 Arch bluetoothd[1642]: Failed to add UUID: Authentication Failed (0x05)
```
So the pulseaudio wouldn't autostart?

darkhz, 2024-03-08:
Interesting, but it seems that you might have to reauthorize a certain service again. Delete the device, and re-add and re-pair it. Then, if you see any "Authorize service" prompts, press "a" to always accept.

5huanghuai, 2024-03-08:
> I think that i know where is the problem. This is the log
>
> ```shell
> ● bluetooth.service - Bluetooth service
>      Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; preset: [...]

OK , I can connect device but can't hear any voice [cry]

darkhz, 2024-03-08:
This is most likely a bluez/pulseaudio issue.

This issue will be closed.

closed 2024-03-08 by darkhz