bluetuith-org/bluetuith #56: Quit application with lowercase Q

graves501, 2024-03-05:
Hi!

First of all thanks for creating and maintaining this project. I use this for my Sway setup and it works great, even better than the X11 GUIs in my opinion!

I was just wondering if it's possible to set the exit/quit keybinding to lowercase Q instead of capital Q because it would be more consistent with other TUI applications like `pulsemixer`. 

Thanks for considering!

Have a nice day :)

darkhz, 2024-03-05:
Keybindings are modifiable, so in your config file (check `bluetuith --help` for the location of your config file) you could add:
```
keybindings: {
      Quit: q
}
```

graves501, 2024-03-05:
Oh! I'm sorry, I should've checked the documentation first. Usually I do that, but I didn't do it this time :sweat_smile: 
It's great that you allow custom keybindings, that's amazing!
And thank you for the quick reply, very much appreciated!

I will jut close this issue then. Thanks again for your wonderful work :grin:

closed 2024-03-05 by graves501