bluetuith-org/bluetuith #58: Using inexistent commit for github.com/darkhz/tview

WhyNotHugo, 2024-03-26:
Current `master` points to a commit in `tview` which does not exist upstream:

```console
> go run ./main.go 
go: downloading github.com/darkhz/tview v0.0.0-20240130154506-5ca91a0d158f
ui/adapter.go:12:2: github.com/darkhz/tview@v0.0.0-20240130154506-5ca91a0d158f: invalid version: unknown revision 5ca91a0d158f
```

```console
> git clone https://github.com/darkhz/tview
> cd tview
> git show 
fatal: ambiguous argument '5ca91a0d158f': unknown revision or path not in the working tree.
Use '--' to separate paths from revisions, like this:
'git <command> [<revision>...] -- [<file>...]'
```

GitHub _does_ show this commit, but it seems to belong only in a fork: https://github.com/darkhz/tview/commit/5ca91a0d158f

Using `github.com/darkhz/tview v0.0.0-20240308094543-6078a888ff79` instead seems to work.

darkhz, 2024-03-26:
Try using the pre built binaries for now, of possible. I have to update the sources, which will be done by this week or the next.

WhyNotHugo, 2024-03-26:
Sure, deleting the entry from `go.sum` and adding the above version worked.

I had to build from source to use a patched dependency due to https://github.com/mafik/pulseaudio/pull/4

WhyNotHugo, 2026-02-06:
This is still an issue trying to build from source. Try cleaning your cache and building with any caching proxy, and build still fails:

    go clean -modcache
    GOPROXY=direct go build .

I tried removing the package from `go.mod` , but `go get` fails with:

```
go: github.com/darkhz/tview@upgrade (v0.0.0-20260113172741-09e574951e58) requires github.com/darkhz/tview@v0.0.0-20260113172741-09e574951e58: parsing go.mod:
	module declares its path as: github.com/rivo/tview
	        but was required as: github.com/darkhz/tview
```