bluetuith-org/bluetuith #59: Cannot connect bluetooth mouse

WhyNotHugo, 2024-04-01:
I'm having issues connecting an MX Master 3B to my host.

## Scenario A

1. Set mouse to pairing mode
2. Open bluetuith
3. Press `s` to scan
4. Select the mouse from bluetuith's UI, `c` to connect

Mouse connects, and immediately disconnects.

## Reset

Select the device, remove it from the list, power cycle it and put it in pairing mode again.

## Scenario B

1. Set mouse to pairing mode
2. Open bluetuith
3. Press `s` to scan
4. Select the mouse from bluetuith's UI, press `t` to trust and press `c` to connect

Mouse connects, and immediately disconnects, the reconnected again and disconnects, ad infinitum.

---

I see this in the logs:

    [Apr 01 17:33:29] daemon bluetoothd[4180]: src/service.c:service_accept() input-hog profile accept failed for XX:XX:XX:XX:XX:XX 

I think that the mouse sends some kind of pairing request/code. I'm not sure how to debug this further.

darkhz, 2024-04-01:
Can you try with any other Bluetooth manager, and see if it pairs? And if it does, can you send a snippet of the logs for that particular device as well?

WhyNotHugo, 2024-04-02:
I don't think that bluetuith can do anything about unloaded kernel modules, so closing.