bluetuith-org/bluetuith #67: Quit keybinding and documentation do not match

Jazztache, 2024-09-03:
Hello!

In Bluetuith on Arch Linux (AUR), the keybinding for quitting is captial 'Q', not lowercase 'q'.
The documentation says it uses the lowercase q keybind to quit, however.
https://darkhz.github.io/bluetuith/Usage/Keybindings.html

I am not sure if the packager messed something up or if the keybindings in the repo itself are misconfigured.
If the binding could be updated to lowercase 'q' instead of capital 'Q', that would be nice.

Thankyou!

soenkehahn, 2024-12-16:
Personally I also think that uppercase `Q` for quitting is bad UI. I'm very used to other TUIs that use lowercase `q` for quitting (e.g. `top`, `htop` & `less`, but there's probably many more). It'd be sad if I had to remember that `bluetuith` uses a different shortcut.

darkhz, 2024-12-17:
As a workaround, for now, use the config file to register the 'Quit' keybinding to 'q'.

View the [configuration docs](https://darkhz.github.io/bluetuith/Configuration.html) here.

I am open to changing the keybinding, the reason 'Q' was used as the quit keybinding initially was to prevent inadvertent exiting of the application.

Configuration:
```hjson
{
keybindings: {
Quit: q
}
}
```