bluetuith-org/bluetuith #69: help does not show 'b' binding for toggling blocked status

aspiers, 2025-03-31:
With 0.2.3, I noticed that one of my devices was showing as blocked.  I couldn't see a key binding in the help to toggle that, but I took a guess that it might be `b`, and sure enough, that worked.  So it looks like an extra line should be added here:

https://github.com/darkhz/bluetuith/blob/2a02b2c397c8515ffeddc32a97d5d25a58c7f766/ui/help.go#L43-L45

aspiers, 2025-03-31:
After looking at the source,, I see it's already in the device menu:

https://github.com/darkhz/bluetuith/blob/2a02b2c397c8515ffeddc32a97d5d25a58c7f766/ui/menubar.go#L105-L109

so it seems I was right that just the help needs updating.

darkhz, 2025-04-01:
Thank you for noticing this. I will update the help menu code soon. Just to confirm, you can see the option in the menubar right?

aspiers, 2025-04-01:
Correct!

darkhz, 2025-04-01:
Okay, noted.