bluetuith-org/bluetuith #73: Version 0.2.5 doesn’t seem to read the configuration file correctly

pdonadeo, 2025-08-22:
I just tested version 0.2.5, and it seems it’s not reading the configuration file. In my file I have these bindings:

```
{
  adapter: "hci0"
  keybindings: {
    Menu: F9
    Quit: q
  }
}
```

If I press `q` (lowercase), nothing happens, while if I press `Q` (uppercase), the program exits. But `Q` is the default binding, so it looks like the file is not being read or processed correctly.

closed 2025-08-22 by darkhz

darkhz, 2025-08-22:
No, it's `bluetuith.conf`. I tested it and it worked for me, I could check again though.

darkhz, 2025-08-22:
Yes it would do that, apparently it creates that file automatically when attempting to port any old configuration to the new format. That has to be fixed, feel free to open another issue. This issue will remain closed however, since it was related to configuration parsing.

pdonadeo, 2025-08-22:
Thanks @darkhz.